nefarius/BthPS3 #1: IRQL NOT LESS OR EQUAL

twindragon6, 2021-01-12:
[logs.zip](https://github.com/ViGEm/BthPS3/files/5800114/logs.zip)

Cornysz, 2021-01-12:
BSOD? I got this BSOD yesterday with same error code too. Got it after stopping shibari service with Control + C. Already reported to Nefarius via discord.

localhost127001, 2021-01-12:
Same here. Two cases IRQL NOT LESS OR EQUAL and one  Kernel Auto Boost Lock Acquisition With Raised IRQL. I don't know is it related or not

nefarius, 2021-01-12:
Thanks but crucial information is lacking. What driver version? What OS version? What architecture? Happens when doing what exactly? Info, info, info!

nefarius, 2021-01-12:
A crash creates a memory dump in `C:\Windows\memory.dmp` right after the crash reboot. I'd need that file (probably a few hundred megs so compress it= over a confidential channel - preferably on the Discord - as it can contain sensitive information and must not be shared in public.

nefarius, 2021-01-14:
Check out v1.3.72, should be fixed.

twindragon6, 2021-01-17:
Fixed in V1.3.72

closed 2021-01-17 by twindragon6