nefarius/BthPS3 #2: KERNEL AUTO BOOST LOCK ACQUISITION WITH RAISED IRQL [BSOD]

Cornysz, 2021-01-12:
BSOD after stopped Shibari, this is my second time appear BSOD after stopping Shibari service.
Any driver information/version should I attach? Thank you in advance

[Raised IRQL.zip](https://github.com/ViGEm/BthPS3/files/5803602/Raised.IRQL.zip)

NB : In the first BSOD, the stop code is IRQL NOT LESS OR EQUAL, I can't find the log.

nefarius, 2021-01-12:
There is no log. A crash creates a memory dump in `C:\Windows\memory.dmp` right after the crash reboot. I'd need that file (probably a few hundred megs so compress it= over a confidential channel - preferably on the Discord - as it can contain sensitive information and must not be shared in public.

nefarius, 2021-01-12:
Since and with which version does it happen?

Cornysz, 2021-01-13:
> Since and with which version does it happen?

I got that error since I updated newest version of BthPS3 (v1.3.65), before that my laptop never encountered by BSOD.
BTW I've sent you friend request, for sending memory.dmp via direct message (discord)

nefarius, 2021-01-13:
Thanks, I found the issue and am providing a fix soon.

nefarius, 2021-01-14:
This has been fixed with v1.3.72

closed 2021-01-14 by nefarius