nefarius/BthPS3 #4: Cannot get working with windows 10 release 1507

sch0009, 2021-02-11:
Hey there, love the functionality with this new software however I cannot seem to get working on a laptop which I have on windows 10 1507. If i update to newer windows it installs however if not I get an error stating that there was an error with the windows installer package. Any ideas? I also run into issues with ViGEmBusSetup_x64 where the device is not recognised after installation. Any help would be much appreciated as I need to stay on this build of windows for driver compatibility reasons. Thanks!

nefarius, 2021-02-11:
Heya,

out of the blue I can't tell you much except that it's quite an old build and I don't even remember if I ever tested one of my drivers on a build this outdated, probably not. 

I can't really tell you anything useful if you don't share way more information on what *exactly* is not working with what drivers versions etc.

nefarius, 2021-02-11:
Hm, the currently latest BthPS3 build depends on KMDF 1.15 which should be compatible with Windows 10, version 1507 (Threshold 1) ([see here](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/kmdf-version-history)).

I don't know if **Threshold 1** is *the* version you currently run, [according to Wikipedia](https://en.wikipedia.org/wiki/Windows_10_version_history_(version_1507)) it's just a codename so should be fine but I'm just guessing, I have no experience with that build whatsoever. My previous questions/statements still stand, need way more information.

sch0009, 2021-02-11:
Thanks very much for the quick response! I have uploaded an image to show the exact error I am getting. I am on a fresh install of win 10 1507 (Threshold 1 as you said) which has no updates at all (when I update to later windows feature releases it messes with some audio drivers for old devices I have so I cannot upgrade unfortunately). Let me know what other additional info you need (also if you have time to look into this of course) and I can happily provide. 

I have also looked into all options around troubleshooting this error such as restarting msiexec.exe, running as admin, running in powershell and command prompt with admin priveledges and a few other things but no sucess unfortunately. 

[https://ibb.co/hRtRHJ6](url)

nefarius, 2021-02-11:
You can directly paste images into responses here btw, no need for an extra image hosting service 😉 

Oh, I see! That error means it can't - for some reason - load the DLL that has the driver installation logic. I've seen it before on users trying to launch it on Windows 7 but so far I have no idea why this happens; my guess is it has something to do with the way the DLL is built/linked on latest Visual Studio 2019 and some dependencies it expects are missing.

So, I now know what's wrong, but not really why I'm afraid 😅 I see if I can find some clues in web searches.

nefarius, 2021-02-11:
Alright, I have an idea, it involves a few manual steps I'll outline:

- Download and extract [BthPS3CustomActions_x64.zip](https://github.com/ViGEm/BthPS3/files/5966814/BthPS3CustomActions_x64.zip)
- Get the [Dependencies tool](https://github.com/lucasg/Dependencies), download and extract somewhere
- Start the `DependenciesGui.exe`
- Go to File / Open... and navigate to the previously extracted `BthPS3CustomActions.dll`
- Share a screenshot of what the tool displays

Thanks

sch0009, 2021-02-12:
Thanks for the suggestion - see below for a screenshot of what the tool is showing. Interestingly enough threre are a few DLL's in there which are red which I am guessing what is potentially causing the issue. No idea how to solve though. 

[image]

sch0009, 2021-02-12:
Also when I go into the msi installation log which is created in the app data temp folder I can see the following:

"Error 1723. There is a problem with this Windows Installer package. A DLL required for this install to complete could not be run. Contact your support personnel or package vendor.  Action CheckHostRadioPresence, entry: CheckHostRadioPresence, library: C:\Temp\MSIB134.tmp 
=== Logging stopped: 12/02/2021  16:57:06 ==="

Not sure whether this has anything to do with bluetooth compatibility. I have an integrated bluetooth device as below. It definitely works as when I tested on an updated version of windows the installation went through it just doesnt seem to want to go on this 1507 build. 

[image]

Thanks again for all your help!

nefarius, 2021-02-12:
Interesting, the MSI one I have on my system as well, so we might ignore that for now. Why it is complaining for `newdev.dll` is surprising though, that is included since Vista according to MS Docs. Can you expand that tree and share the view again? Thanks.

sch0009, 2021-02-12:
If I expand it out I can see the following:

[image]

If I expand others out I can also see other errors which seem to relate to shell32

[image]

Thanks again

nefarius, 2021-02-12:
This is a weird one. I unfortunately don't really have an idea ready what to recommend next so I'll just download 1507 myself (thankfully Archive.org has a mirror copy of the ISO) and see if I can replicate the issue.

nefarius, 2021-02-12:
*Back to the future...*

[image]

nefarius, 2021-02-12:
[image]

sch0009, 2021-02-13:
Haha 'back to the future'. I wish I didnt have to live in the future however the audio interface I have just doesnt want to work with newer versions of windows. Its a shame that the manufacturer wont update their drivers. Anyway I really appreciate you going to the effort to look into this and see whats happening with the build. I tried pretty much everything I could find last night and still cannot get it to work from my end. Thanks!

nefarius, 2021-02-15:
👀 

[image]

nefarius, 2021-02-15:
Haha, I found the issue, it's this stupid API [`PathCchCombine`](https://docs.microsoft.com/en-us/windows/win32/api/pathcch/nf-pathcch-pathcchcombine) to combine paths, it's only available on Windows 8 and apparently later variations of 10 *sigh*

I'll replace it, only 3 lines using that and see if that is enough.

nefarius, 2021-02-15:
Oh you gotta be kidding me, it's still not working! 😲

sch0009, 2021-02-15:
Oh man I actually woke up at 5am last night and saw your first message and was very excited that it seemed to be fixed (I am in Australia). Sorry to hear that it did not work in the end I know the feeling. As always thanks again for continuing to try to resolve this.

nefarius, 2021-02-15:
I fixed it half an hour ago, for real this time 😁 Switched from static linking to dynamically loading the troubled libraries in question and now it works like a charm! A new problem has surfaced now though; driver signature seems to be not working on that particular build of Windows 10 😦 

[image]

Which basically means the solution can get installed but the kernel refuses to load the drivers rendering the whole operation useless 😞

sch0009, 2021-02-16:
Ah that really sucks. Just when you think you find the solution another issue comes up. Looks like some good progress though so well done on fixing the initial issue at least!

nefarius, 2021-02-16:
Yeah discovering that fix was quite important though, benefits other projects as well.

Did a bit more digging and unfortunately 1507 is one of those editions where the signing policies were still very messy, like the transition from Windows 7 through 8.x and to the strategy of 10 now since 1607. Tomorrow (today) I can do a bit more testing to see if there's any "loophole" or other official fix available.

Off-topic: are you sure you *need* to be on such an old build? Like, what exactly happens with your mentioned audio gear on newer releases?

Cheers

nefarius, 2021-02-16:
Hm, looking at `C:\Windows\inf\setupapi.dev.log` I can't find any driver installation errors... Maybe it's VMware? I can shoot you the MSI to test later on your machine, to rule out the issue is on my end.

sch0009, 2021-02-16:
Ok that would be great heppy to test out the MSI also happy to test out any potential fixes as I have been reading up on that specific code error 39 and there are some suggested workarounds I can try also. 

Yes unfortunately its the only build which works for my mixer (Rane MP2015 mixer). For some reason any time I update windows I get horrible audio stutter. I have tried multiple versions of windows, with multiple fixes each time and to be honest have spent a  few months trying to diagnose but for the life of me cannot come up with anything. I have contacted the developers to see if they are ever intending to update their drivers but have never got a response unfortunately. 

I am glad the fix is at least beneficial for you and that it helps out with other projects!

nefarius, 2021-02-16:
Alright, [be my guest to test this release](https://downloads.vigem.org/projects/BthPS3/testing/1.3.118/) on your end! 😀

sch0009, 2021-02-16:
Thanks for sending through - unfortunately I am getting the same error on my end. I will look up and see if there is anything I can find today. Hey at least it installed though :)

[image]

[image]

sch0009, 2021-02-17:
Its actually interesting as this is the same error which I came across when I installed the latest ViGEm Bus Driver (1.17.333). If I installed that version on windows 1507 it had the same code 39 error above and the yellow triangle on the system device. However when I tried ViGEmBus Setup (1.16.116) it installed correctly.

So at the moment I am using ViGEmBus version 1.16.116 which works on my OS build and I have tested everything works with Shibari via USB. I wonder if there are any learnings which can be used as to what canged from ViGEmBUS version 1.16.116 -> 1.17.333 which prevented installation on win10 1507? Or maybe even the potential of older BTHPS3 packages which may have been compatible. The two which are up here did not work however I can see previous msi executables for previous versions were removed. 

Just some thoughts I dont have much else - I did try a few things from guides I found online around tweaking registry to remove upper and lower values and reinstalling the drivers however still no luck. 

Again thanks very much for continuing to try to troubleshoot this.

sch0009, 2021-02-17:
If I look at event logs I can also see the following - not sure if this helps:

[image]

[image]

nefarius, 2021-02-17:
*Sigh* this is unfixable without going through WHQL I'*m afraid, attestation signing seems to not work for Windows 10 editions this old. This process is a huge PITA and takes hours to do and needs to be done again every time even one bit in the derivers change so you can imagine I'm not really a fan of that. So for now I'm afraid this case is unresolvable, sorry mate 😬

sch0009, 2021-02-17:
That does sound like a huge PITA so definitely understand leaving it in the unresolved bucket. I am going to try some additinal builds of windows 10 and see if any others play nicely with my audio driver (even though I have tried about 5-6 dif versions already). I dont think I tried 1607 so maybe there will be some hope there (well thats if the BTHPS3 drivers work on that version haha. I will hold my breath I guess. Thanks again for all the trouble that you went through diagnosting this over the past couple of days its really appreciated despite the fact that we didnt get to a resolution. Take care and thanks again.

nefarius, 2021-02-17:
Good luck on your challenge! Unfortunately WHQL testing literally requires me to set up a few PCs, install a server and software for the test kit, prepare the machines, pray that it even works, submit the results to Microsoft, pray it gets accepted and no errors happened in submission and then see *if* it works for you. That is a task for 6 to 12 hours of work without being certain it even helps so just to give context on why I'm not a fan to just give into that mess 😬

nefarius, 2021-02-17:
Actually my theory is wrong, for the record. The filter driver loads fine and even produces trace output. The profile driver is complaining though. Have to admit at this point in time I have no idea why though. Will keep this issue open, maybe some Eureka moment arrives sooner or later 😉

sch0009, 2021-02-18:
No worries I will keep you updated on my end also. As an FYI I have ruled out a few other OS builds which do not work - 1511, 1603, 1607 all run into the same errors. I am just testing 1709 now and the drivers work - just need to retest my audio drivers (takes at least a few hours to see stability). Fingers crossed. If this doesnt work I guess I am out of luck as I have tested everything post 1709. Fingers crossed for a eureka moment who knows!

nefarius, 2021-02-18:
Thanks for reporting; this is really an interesting case that thankfully seems to have nothing to do with the signature; I just tested v1.2.3 out of curiosity and it works!

[image]

So switching from KMDF 1.11 to 1.13 introduced an issue that totally prohibits loading the profile driver but not the filter, as that one still uses KMDF 1.11... Puzzle time!

sch0009, 2021-02-18:
Wow that's awesome so you did have a eureka moment? Would that version work for me or is it incompatible with Shibari and the other apps required for Bluetooth? So cool that you figured that out. Would be happy to give the older ones a try. Giving me hope. 

Interestingly enough 1709 worked with the driver installation however my audio drivers but out at that point. Something must have changed between 1607 and 1703/1709 to cause issues with my audio driver and make your ones work haha. No happy medium for me.

nefarius, 2021-02-18:
> Wow that's awesome so you did have a eureka moment? Would that version work for me or is it incompatible with Shibari and the other apps required for Bluetooth? So cool that you figured that out. Would be happy to give the older ones a try. Giving me hope.

Never say never, I just fixed it with the latest greatest version 🥳 

[image]

I recently switched from KMDF 1.11 (Windows 7 minimum) to v1.13 (Windows 8.1 minimum) to support features I need in the [upcoming Shibari successor](https://vigem.org/projects/DsHidMini/) and accidentally set the OS to build for to 10, this seems to cause linker issues with KMDF 1.13, so I set it correctly and it loads in my test VM 😎 

[image]

I'll prepare a production release and throw the link your way to test once ready. Thanks for sticking with me in this adventure 😁 

> Interestingly enough 1709 worked with the driver installation however my audio drivers but out at that point. Something must have changed between 1607 and 1703/1709 to cause issues with my audio driver and make your ones work haha. No happy medium for me.

I'm really surprised it gives you that much headache; ever since Vista driver compatibility has greatly improved (*If* you build your stuff correct like demonstrated by yours truly haha).

sch0009, 2021-02-18:
I have to say I am very impressed with the dedication on this it's pretty amazing that you solved it. Literally just jumped up in the air and said hooray. Well done on identifying the issue and resolving it. Can't wait to test it out. 

It also means I can stop trying to troubleshoot my audio driver (which as I mentioned have probably spent 60+ hours doing). I don't know if it's driver incompatibility with new versions of windows or the way the OS is actually built but it never seems to work on anything other than 1507. So strange as you would think newer versions would be okay. It's one of those things where you think it's fixed but then 20 minutes into mixing some music you get a big audio stutter. Running latency on on it indicated that it's to do with windows driver foundation however it does not pinpoint anything unfortunately. 

Anyway thanks again for pushing through with this so happy with the outcome and really appreciate the awesome work.

nefarius, 2021-02-18:
> It also means I can stop trying to troubleshoot my audio driver (which as I mentioned have probably spent 60+ hours doing). I don't know if it's driver incompatibility with new versions of windows [...]

Have you had a go with this? https://resplendence.com/latencymon

sch0009, 2021-02-18:
Yeah I have run it extensively but not too helpful to be honest for the most part. It's one of those things where you try 100 things and each time you think it's working until it stuffs up. And considering 20+ minutes each time testing to detect the stutter it gets quite time consuming going through a process of elimination. Who knows it could be something simple with a conflict but I just can't track it down. 

Also just as a note the newest ViGEm Bus Driver (1.17.333) also did a similar thing with the yellow icon on the driver (think it was the same code). It installed but didn't work. Luckily 16.16.116 worked. The reason I mention it is I wonder whether it's the same thing going on with that driver that you picked up here.

nefarius, 2021-02-18:
> Yeah I have run it extensively but not too helpful to be honest for the most part. It's one of those things where you try 100 things and each time you think it's working until it stuffs up. And [...]

If it takes some time to build up until the problem manifests itself it could be a memory leak; if you still have a "problematic" Windows installation at hand a primitive but easy check is to have a look at Task Manager / Memory / Non-paged pool, if that rises continuously to over a gigabyte and doesn't come down again it's an indicator:

[image]

> Also just as a note the newest ViGEm Bus Driver (1.17.333) also did a similar thing with the yellow icon on the driver (think it was the same code). It installed but didn't work. Luckily 16.16.116 [...]

Can you post a screenshot with the details? Could very well be a similar issue! Most of my "test flock" are on bleeding edge recent builds 😉 

Cheers

sch0009, 2021-02-18:
Thanks for the tip I will take a look shortly as I am still on 1709 which has the stutter also so will keep my eye on the memory section in task manager. 

When I have done this (and likely unsuccessful) then I will roll back to 1507 and send across the screenshot :)

nefarius, 2021-02-18:
> Thanks for the tip I will take a look shortly as I am still on 1709 which has the stutter also so will keep my eye on the memory section in task manager.

I had a brief look at the product page and knowing a bit about audio gear I assume it was quite the investment so I'd recommend to hammer the customer support a bit more aggressively, in this day and age it really is no excuse anymore to let hardware become obsolete by providing cr*p software 😉

sch0009, 2021-02-18:
Haha yeah I have been on the warpath for the past few months trying their Facebook support, Facebook pages, official webpage forums and support email however still nothing. It's very disappointing because as you rightly said it was a significant investment for me.

nefarius, 2021-02-18:
I feel you 😦 Last time I rang up customer support for some audio gear was like 6 or 7 years ago. Wanted repair manuals for a PA amp. They straight up lied to my face they don't have any schematics but I can send it to them for repair for coin... OK so how will you diagnose and repair it without schematics, you geniuses... So I hung up, reversed and fixed it myself 😎 

Back to topic: [setup ready for test](https://downloads.vigem.org/projects/BthPS3/testing/1.3.125/)!

Cheers

sch0009, 2021-02-18:
Hey so guess what it all worked! I just connected everything up with shibari installed as a service and got the bluetooth driver installed sucessfully this time. Very impressed with your quick responses and very happy that you had a eureka moment in order to solve the issue. Here is a screenshot of the driver now installed. 

[image]

Also I checked the ViGEm Bus Driver (1.17.333) and it is showing the same error code 39 as per this issue so it may fall into the same bucket for the resolution. 
[image]

As mentioned before when I roll that back to ViGEm Bus Driver (16.16.116) the driver loads sucessfully. I have added a capture of what that looks like in device manager also. 
[image]

You are a legend and should be proud of your work. Amazing support on the topic. Oh and hey maybe there will be others in my group of not being able to upgrade their OS who will benefit from this fix you have put in!

nefarius, 2021-02-18:
> You are a legend and should be proud of your work. Amazing support on the topic. Oh and hey maybe there will be others in my group of not being able to upgrade their OS who will benefit from this fix you have put in!

❤️ Thank you kindly, this kind of feedback keeps the drive to tinker and fix alive 😁 

Thanks for investing the time, reaching out and being available for feedback and tests, it's the only way these kind of issues surface and can get resolved. I'll cook up an official release as we speak.

Hope you get the audio issue resolved as well, sooner or later the need to upgrade will creep around the corner 😬 

Cheers

mrpenguinb, 2021-02-22:
> Yes unfortunately its the only build which works for my mixer (Rane MP2015 mixer). For some reason any time I update windows I get horrible audio stutter. I have tried multiple versions of windows, [...]

>I am just testing 1709 now and the drivers work - just need to retest my audio drivers (takes at least a few hours to see stability). 
It sounds most likely that Windows 10 overriding the drivers for your mixer.
You either have to re-install the drivers made by Rane each time there is an update to Windows, or disable Windows 10 from updating your device drivers automatically in the background : https://www.windowscentral.com/how-disable-automatic-driver-updates-windows-10

By the way Nefarious, the new installer for BthPS3 works significantly better than the old installer you had.
Now I can actually use BthPS3! 🎉

nefarius, 2021-02-22:
> By the way Nefarious, the new installer for BthPS3 works significantly better than the old installer you had.
> Now I can actually use BthPS3! 🎉

Glad to hear! I've basically rewritten the whole installer logic and switched from a commercial solution to a completely free and open setup maker ([WiX](https://wixtoolset.org/)).

closed 2021-03-25 by nefarius

nefarius, 2023-06-14:
This chapter is once again not done; I found [two](https://fosstodon.org/@Nefarius/110538974979824362) more [reasons](https://fosstodon.org/@Nefarius/110542714882128138) why latest BthPS3 doesn't like to load on anything lower than 1803 and I am confident it is fixable, stay tuned!