nefarius/BthPS3 #8: Qualcomm Atheros AR3012 Not working (Integrated laptop BT adapter)

AshIndigo, 2021-03-12:
I just installed this to try and get BT working with my controller, but I'm getting this error. Is it just not compatible and I'm out of luck or is there something I can do?

> This device cannot start. (Code 10)
> 
> STATUS_DEVICE_POWER_FAILURE

nefarius, 2021-03-12:
Please share the Hardware ID and HCI/LMP version (see README and [here](https://vigem.org/projects/BthPS3/Compatible-Bluetooth-Devices/)

AshIndigo, 2021-03-12:
HCI 6.4354 / LMP 6.1
Hardware ID: USB\VID_0CF3&PID_3004 and USB\VID_0CF3&PID_3004&REV_0001

nefarius, 2021-03-12:
[Please exercise this](https://vigem.org/projects/BthPS3/Debugging-the-drivers/) by disabling your Bluetooth adapter, start the trace, enable it so the error gets captured, stop the trace and attach it here.

AshIndigo, 2021-03-12:
[BthPS3.zip](https://github.com/ViGEm/BthPS3/files/6130556/BthPS3.zip)

Sorry for the delay, I've attached the etl file inside of the zip

nefarius, 2021-03-12:
No worries. Interesting, never seen before 🤔 

```
 [BthPS3_RegisterL2CAPServer] ERROR:BRB_REGISTER_PSM failed with status STATUS_DEVICE_BUSY (0x80000011)
```

Does it work if you if you disable and enable the `Nefarius Bluetooth PS Enumerator` device?

[image]

AshIndigo, 2021-03-12:
Just toggled it on and off by disabling and enabling, and still got the same message as before.

nefarius, 2021-03-12:
Damn, uh, good question then 😬

nefarius, 2021-03-25:
Yeah mate I'm afraid I have no idea on this one...

closed 2021-07-09 by nefarius