nefarius/BthPS3 #10: USB\VID_0A12&PID_0001 Issue (USB BT Adapter)

AshIndigo, 2021-04-01:
Hi it's me again, so I ended up ordering another Bluetooth adapter from Amazon that I was sure would work with this. 
Unsurprisingly with my luck, it did not end up working. (Code 10, same old stuff)

I feel like its due to the fact that I have two BT adapters now, my built in one and this usb one I got, but I don't know how to confirm that theory.

I attached the ETL file below, hope it helps.

[BthPS3.zip](https://github.com/ViGEm/BthPS3/files/6242578/BthPS3.zip)

nefarius, 2021-04-01:
You can not have 2 active Bluetooth host radios in Windows active at the same time, you need to disable the built-in device in Device Manager and make sure that only the new USB dongle is active.

AshIndigo, 2021-04-01:
Admttedly, I did already end up doing that before making this issue and ended up forgetting to mention it.
[image]

AshIndigo, 2021-04-16:
Good news, I got it fixed
Bad news, The solution makes me look like an idiot.

Ended up swapping the USB port my new USB adapter was on and it started working. So might have just been a hardware issue considering this laptop is from 2014.

Guess this can be a lesson to anyone else having odd issues

closed 2021-04-16 by AshIndigo

nefarius, 2021-04-16:
Thanks for reporting back. Yeah, USB can be weird. Especially the transition from USB 2.0 to 3.x ports, which *should* be completely backwards compatible *but* some manufacturer chipsets and/or drivers can misbehave for unknown reasons. Computers!