nefarius/BthPS3 #13: Windows on ARM, Support?

sidd-kishan, 2021-05-30:
Please let us know when can we have an ARM64 version for Windows on ARM OS.

nefarius, 2021-05-30:
Never, as long as I'm doing this alone and fuelled by a donation every now and then, I don't own any ARM hardware and therefore have no interest or means to port/compile/support for that platform.

sidd-kishan, 2021-05-30:
We at Windows on Rasberry Pi community have ported windows on arm to Rasberry pi 4 b+ we can help you with the testing. If your efforts are good we can try to start a crowdfunding campaign for you.

nefarius, 2021-05-31:
I think my countless past efforts speak for themselves 😉 is there more I can read about this community and said efforts?

nefarius, 2022-03-24:
Good news to the two people possibly reading this: I got my first ARM64 device and will tinker with it soon 😄

nefarius, 2022-08-16:
So the ARM64 driver builds work fine but the distribution of a working setup will be tricky [as it appears that no official stable WiX with ARM64 support exists yet](https://github.com/wixtoolset/issues/issues/6269) 🙁 

So we either wait or switch to Advanced Installer 🤔

nefarius, 2022-08-16:
Ah, digging a bit through the related issues it's claimed that [WiX 3.14](https://wixtoolset.org/releases/v3-14-0-712/) (a pre-release?) offers ARM64 support, will try to install that and see what happens 🤞

nefarius, 2022-08-16:
> Ah, digging a bit through the related issues it's claimed that [WiX 3.14](https://wixtoolset.org/releases/v3-14-0-712/) (a pre-release?) offers ARM64 support, will try to install that and see what happens 🤞

Outdated link, [this release](https://wixtoolset.org/releases/v3-14-0-6526/) ships with ARM64 SDK libs 👀

nefarius, 2022-08-18:
Currently battling with CI to build the ARM64 components the setup depends on 🤞

nefarius, 2022-08-18:
Signing...

[image]

nefarius, 2022-08-19:
[Release](https://github.com/ViGEm/BthPS3/releases/tag/v2.5.148.0) 🥳

closed 2022-08-19 by nefarius