nefarius/BthPS3 #14: Auto-pairing in other paired devices doesn't work (CSR USB dongle)

richterdeveloper, 2021-06-22:
Since I installed BthPS3, I need to re-pairing my 8bitdo controller (I need to do all process in the Windows Device Configuration panel every time after disconnect the controller). I don't know if this is an issue with my CSR dongle or is an issue in BthPS3.

Thanks!!

nefarius, 2021-06-22:
[See this piece of documentation](https://vigem.org/projects/BthPS3/Driver-Configuration-Utility-Explained/#filter-driver-settings), I'm afraid this is a side-effect of the overall design (I highly recommend to read all the provided docs carefully) since the filter probably interferes with the controller of yours. Try turning it off (explained in the docs) and if that doesn't help, I'm afraid the only solution is to not use the software in conjunction with your controller in question.

richterdeveloper, 2021-06-23:
> [See this piece of documentation](https://vigem.org/projects/BthPS3/Driver-Configuration-Utility-Explained/#filter-driver-settings), I'm afraid this is a side-effect of the overall design (I highly [...]

This solution is working flawless. Thanks!!

closed 2021-06-23 by nefarius