nefarius/BthPS3 #26: Driver not retained when migrating from Windows 11 22000 to 22449 and likely all build upgrades

dxgldotorg, 2021-09-03:
- [x] I have searched open and closed issues for duplicates
- [x] I am submitting a bug report for existing functionality that does not work as intended
- [x] I have read https://vigem.org/Community-Support/
- [x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
Driver for BthDS3 is lost when upgrading from Windows 11 build 22000 to 22449.

### Steps to reproduce
- Install Windows 11 from ISO
- Install DsHidMini and BthPS3
- Enroll in the Dev channel and install the new build

**Actual result:** Controller no longer connects, Nefarius Bluetooth PS Enumerator is no longer present in Device Manager.
**Expected result:** Driver should be still installed.

### Machine info

**CPU-Architecture:** x64
**Windows version:** Windows 11 builds 22000 and 22449
**Software/driver version(s):** 2.1.153

### Any other helpful information

The MSI "Repair" option is not available to reinstall the drivers in one step; have to uninstall, reboot, and install again to get the drivers back.

dxgldotorg, 2021-09-12:
Last week's Dev build killed it again; it appears the migration system causes the driver to get uninstalled.  Perhaps something to do with the method used to install it?

nefarius, 2021-09-12:
I'm doing nothing special that hasn't been done for like 20+ years now and I am tired running after things Microsoft breaks 🙂

nefarius, 2022-07-20:
Closing as Windows 11 doing Windows 11 silliness is outside of my scope.

closed 2022-07-20 by nefarius