nefarius/BthPS3 #44: BthPS3 installer breaks/uninstall ViGEmBus

Kanuan, 2022-06-05:
Tested on v2.1.153

# This issue is currently a draft

Installing or uninstalling BthPS3 (can't remember right now) causes ViGEmBus to not be detected anymore by applications looking for it. I'm not sure if the driver itself does not get loaded anymore by Windows or if it's loaded but applications can't find it.

Regardless, it's possible to just re-install the ViGEmBus without major problems.

# To-do

Reproduce the issue again to check if the ViGEmBus still appears as installed on windows' Apps and Features list or is present in the Device Manager

nefarius, 2022-06-05:
I had a look at this a few weeks (months?) back but need to investigate again, I think it is due to some reused code/scripts in the setup that accidentally nukes the wrong device.

nefarius, 2022-08-05:
Can we somehow get a playlist on how to exactly reproduce it? It is still happening, observed on the Discord, but with just text search I can't find any references to ViGEmBus and its setup in the BthPS3 installer sources whatsoever 🤔 So best chance might be to reproduce it and log what happens during it with Process Monitor to find the originating process that kicks off the uninstall command.

nyakze, 2022-08-14:
It also seem to break "Hide DS4 Controller" option in DS4Windows, as I can't hide my DS4 anymore, 2 of them appears. Reinstalling or toggling the option doesn't help.

nefarius, 2022-08-14:
> It also seem to break "Hide DS4 Controller" option in DS4Windows, as I can't hide my DS4 anymore, 2 of them appears. Reinstalling or toggling the option doesn't help. Getting: Could not open DS4 90:89:5F:94:7C:65 exclusively. You must quit other applications like UWP apps (Netflix), Steam, Uplay, Edge, NVIDIA IN-GAME before activating the 'Hide DS4 Controller' option

Having a hard time believing that since I see no relation to that feature whatsoever.

nyakze, 2022-08-14:
Could be related to dshidmini, as I install them together. But this problem has been reproduced on 3 completely different systems.

nefarius, 2022-08-14:
> Could be related to dshidmini, as I install them together. But this problem has been reproduced on 3 completely different systems.

DsHidMini currently doesn't ship with an installer and on its own doesn't have any code or logic shipped that interfaces with what you described.

ElHurta, 2022-08-19:
Hi everyone
Same here, I'll try to reproduce my steps

I had already installed DS4Windows in my PC, for my Dualsense Controller:

1. Installed dshidmini_v2.2.282.0 x 64
2. Connected Via USB my 2 Dualshocks 3
3. Connected my Bluethooth device
4. Activated DS4Windows Mode
5. Installed BthPS3 v2.5.148 x64
6. My PC Reboots
7. Opened DS4 Windows and ViGEMUSB is not installed
8. Installs ViGEmBus and everything works fine EXCEPT BthPS3 appears as not installed in my DsHidMini Control
9. I tried again, and checked BthPS3 installation without installing ViGEmBus and still as not installed

nefarius, 2022-08-20:
I can not reproduce it whatsoever. Maybe it's code in DS4Windows? @Kanuan does DS4Windows silently invoke any related setup on initial start?

closed 2022-08-21 by nefarius

nefarius, 2022-08-21:
Goodness, copypasta bug hiding in plain sight for quite some years 🤦 [new release has that fixed](https://github.com/ViGEm/BthPS3/releases/tag/setup-v2.5.152.0)!

commit 82ddf66d (2022-08-21): Fixes #44