nefarius/BthPS3 #49: Broke "DS4Windows" mode in DsHidMini

nefarius, 2022-08-05:
Regression reported by @Kanuan that wireless DS4Windows mode on DsHidMini broke wit the refactoring changes. Looks like the Device ID is different, check new PDO creation code for missing instructions compared to old version.

### New beta

\?\hid#{53f88889-1aaf-4353-a047-556b69ec6da6}&dev&vid_054c&pid_0268#a&38e13f01&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

### Latest Stable

\?\hid#{53f88889-1aaf-4353-a047-556b69ec6da6}&dev&vid_054c&pid_0268&**0498f3913fce**#a&4179438&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

nefarius, 2022-08-13:
@Kanuan is `04:98:F3:91:3F:CE` the MAC address of the controller you tested with? Can you please confirm, thanks.

Kanuan, 2022-08-13:
yes

[image]

nefarius, 2022-08-13:
What an odd issue indeed, I have fixed the Device ID format in my local test build but it still doesn't work 🤔

nefarius, 2022-08-13:
I'll focus on the crash-relevant fixes and then come back to this once I have a new production signed copy I can then debug directly with DS4Windows.

nefarius, 2022-08-15:
Found it. DS4Windows checks if a detected controller is a virtual device to exclude them from detection and it so happens that the current check now also triggers on the DS3:

[image]

Thinking on what's the best way to fix this, IMHO the detection in DS4Windows should be patched since this check method is very brittle to begin with. More to come.

nefarius, 2022-08-25:
Success!

[image]

Preparing PR.

nefarius, 2022-09-10:
The fix has been merged into DS4Windows 🥳 Now we wait for a new release ☕

nefarius, 2022-09-23:
[A fixed version has been released](https://github.com/Ryochan7/DS4Windows/releases/tag/v3.1.6).

closed 2022-09-23 by nefarius