nefarius/BthPS3 #66: BthPS3 Code 31

ELXShadow, 2022-12-23:
- [x] I have searched open and closed issues for duplicates
- [x] I am submitting a bug report for existing functionality that does not work as intended
- [x] I have read https://vigem.org/Community-Support/
- [x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
Like many other people, im getting the Code 31 error message even after running the powershell command in the website. And DsHidMini isn't detecting BthPS3 at all.

### Steps to reproduce
- Install DsHidMini
- Install BthPS3
- Run the powershell command

**Actual result:** No bluetooth detected, Not detected in DsHidMini before running the command. After running the command, Code 31, normal bluetooth works fine (Not BthPS3 tho), still not detected in DsHidMini. And also BthPS3 Config Tool wont launch if Bluetooth is on.

**Expected result:** Without any command or registery edits, Bluetooth and BthPS3 both work fine, and is detected by DsHidMini,

### Screenshots
[image]
[image]
[image]
[image]

### Machine info
**CPU-Architecture:** x64
**Windows version:** Phoenix LiteOS 11 Pro Plus (based on Windows 11 22H2 22621.819)
**Software/driver version(s):** Latest

### Any other helpful information
This also happened with other people, I have no clue what's going on and I've been trying to fix it for weeks. Thanks in advance.

nefarius, 2022-12-23:
This error means the filter driver BthPS3PSM wasn't loaded, which is mandatory for BthPS3 to work.

I can't tell you why, it boils down to broken or weird Windows installations, I can't reproduce it, and I can't offer a fix besides what's already documented.

nefarius, 2022-12-23:
Potentially related to #64

ELXShadow, 2022-12-25:
> it boils down to broken or weird Windows installations

Maybe cuz of the Lite OS?

nefarius, 2022-12-25:
> > it boils down to broken or weird Windows installations
> 
> Maybe cuz of the Lite OS?

IDK what that is supposed to be, Windows 11 is a shameless Beta product anyway. If you further manipulate it I doubt things get any better.

nefarius, 2022-12-25:
Give me a couple of days and I can make a release with more logging added to the filter driver, maybe then we get closer to the culprit.

ELXShadow, 2023-01-21:
> Give me a couple of days and I can make a release with more logging added to the filter driver, maybe then we get closer to the culprit.

Any updates?

nefarius, 2023-01-21:
Had no time yet unfortunately 😕

ELXShadow, 2023-02-12:
This is also happening on debloated Windows 10

yanuarmdhn, 2023-03-21:
https://github.com/ViGEm/BthPS3/discussions/25#discussioncomment-3655329

> I updated VigemBus and BthPS3 to the latest version, my 4.0 dongle kept working normally and my 5.1 dongle now gives me code 10. When I remove the filter, I can at least use the 5.1 dongle with other things, but the "Nefarius Bluetooth PS Enumerator" shows the code 31 =/.
> 
> Dongle 5.1 [image]
> 
> Apparently the only solution is to use another dongle. So, I don't recommend you to buy the Baseus Bluetooth Dongle ba04 5.1 if you want to use BthPS3. (Although I vaguely remember seeing in some video someone using version 5.0 of this dongle working normally.)
> 
> Anyway, thanks for the help!
> 
> Dongle 4.0 [image]

I think this is the problem. Driver BthPS3 Bluetooth Drivers v2.6.174 doesn't support Bluetooth dongle 5.0. I tried to install Bluetooth dongle 5.0,  My OS Windows 10 Latest Update, Dshidmini latest. i already follow all installation and troubleshooting instructions, the results always failed. Maybe the devs will fixed on the next update.

tiag0d, 2023-04-18:
> [#25 (reply in thread)](https://github.com/ViGEm/BthPS3/discussions/25#discussioncomment-3655329)
> 
> > I updated VigemBus and BthPS3 to the latest version, my 4.0 dongle kept working normally and my 5.1 dongle now gives me code 10. When I remove the filter, I can at least use the 5.1 dongle with other things, but the "Nefarius Bluetooth PS Enumerator" shows the code 31 =/.
> > Dongle 5.1 [image]
> > Apparently the only solution is to use another dongle. So, I don't recommend you to buy the Baseus Bluetooth Dongle ba04 5.1 if you want to use BthPS3. (Although I vaguely remember seeing in some video someone using version 5.0 of this dongle working normally.)
> > Anyway, thanks for the help!
> > Dongle 4.0 [image]
> 
> I think this is the problem. Driver BthPS3 Bluetooth Drivers v2.6.174 doesn't support Bluetooth dongle 5.0. I tried to install Bluetooth dongle 5.0, My OS Windows 10 Latest Update, Dshidmini latest. i already follow all installation and troubleshooting instructions, the results always failed. Maybe the devs will fixed on the next update.

I'm having the exact same issue with a bluetooth 5.0 dongle. If this is an issue with the program I hope to see it fixed when possible.

Wiiplay123, 2023-05-05:
This happened to me with all of my USB bluetooth devices including my HTC Vive bluetooth after installing BthPS3, I uninstalled and it kept happening. I reinstalled and it _continued_ happening, but the HTC Vive bluetooth started working again after reinstalling its drivers.
I found out that BthPS3 was actually working with my motherboard's integrated bluetooth so I didn't need that extra USB bluetooth dongle anyway. I have a .etl file if you need it.

I'm on Windows 10 Home Version 21H2, upgraded from Windows 7.

nefarius, 2023-06-16:
#82 might be related.

nefarius, 2023-07-08:
New pre-release is out, please test](https://github.com/ViGEm/BthPS3/releases/tag/setup-v2.7.296)!

nefarius, 2023-07-10:
Tested with a wide range of Windows 10 and 11 versions, I consider this fixed.

closed 2023-07-10 by nefarius

DeadKiller25, 2023-07-16:
Still getting Code 10 on BthPS3 install (both recent versions), and Code 31 on the Nefarius Bluetooth PS Enumerator after running the PowerShell command. I'm running Windows 10 Home - 22H2.

My dongle is a Generic Bluetooth Adapter, it has a v5.0 printed on it, and the HCI and LMP are v10.137. I've tried both SCPToolkit and DshidMini/BthPS3 and never got it to work wirelessly, I think this adapter's simply not gonna do the job :/

nefarius, 2023-07-16:
Welp.

celsosa, 2023-12-10:
I have same problem. I'm using Baseus bluetooth adapter 5.1

DeadKiller25, 2023-12-13:
> I have same problem. I'm using Baseus bluetooth adapter 5.1

You know... I had a old netbook gathering dust and I remembered it had a good bluetooth adapter. Well, I ended up using it's adapter over LAN using [this software](https://www.virtualhere.com). It allows to share over network any USB device, and from my tests, WiFi offers a somewhat crappy experience, though LAN seems just fine :)

ShadowElixir, 2024-05-14:
Still not working with the latest version, I think this only happens with 5.1 bluetooth adapters.

ShadowElixir, 2024-05-14:
@nefarius

nefarius, 2024-05-14:
> @nefarius

No need to ping, I am able to read on my own accord 😛 IDK why this happens, hence can not offer any aid.