nefarius/BthPS3 #72: Trouble uninstalling

tiag0d, 2023-04-18:
Hey I installed the software without running into any error messages and after restarting my computer as requested by the installer my bluetooth dongle wasn't working.

Now I wanted to reinstall the software to see if it fixes but when I hit uninstall I get a error message saying "Bluetooth Host Radio not found. A working Bluetooth stack is required for driver installation or removal."

Is there another way to uninstall this program?

nefarius, 2023-04-18:
https://vigem.org/projects/BthPS3/Frequently-Asked-Questions/

closed 2023-04-18 by nefarius

ellie12134, 2025-03-09:
that link doesnt work anymore, what was even the solution?

nefarius, 2025-03-09:
https:// docs.nefarius.at/projects/BthPS3/Frequently-Asked-Questions/

Not that hard to find.