nefarius/BthPS3 #73: "Enable PSM patching" re-enables itself on reboot

InterClaw, 2023-05-03:
- [x] I have searched open and closed issues for duplicates
- [x] I am submitting a bug report for existing functionality that does not work as intended
- [x] I have read https://vigem.org/Community-Support/
- [x] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description
After having disabled the PSM patching, by unticking the checkbox for it, and restarting the computer, the checkbox has automatically ticked itself again. It doesn't seem like the setting is stored anywhere. I think it's supposed to be stored in the registry on a value called "IsPSMPatchEnabled", but this doesn't seem to get written anywhere.

I've uninstalled and reinstalled the utility, but the problem remains.

Note that disabling the filter does work as intended while keeping the computer running. It's just that it re-enables itself on reboot.

### Steps to reproduce
- Install BthPS3.
- Start the utility.
- Click yes in the elevation prompt.
- Go to tab "Filter Driver Settings".
- Untick "Enable PSM patching.
- Close the utility and restart the computer.
- Open the utility again.

**Actual result:** The checkbox is ticked again. And the filter is active.
**Expected result:** The checkbox should remain unticked and the filter should remain inactive.

### Screenshots

### Machine info
**CPU-Architecture:** x64
**Windows version:** Windows 10 Enterprise 22H2
**Software/driver version(s):** v2.6.174

### Any other helpful information
Here is a discussion of the problem in Discord:
https://discord.com/channels/346756263763378176/1025174973498789918/1099328923738837013

InterClaw, 2023-05-04:
I could try to manually add it to the registry and see if it at least reads it like it’s supposed to. 🙃

Could you confirm where it’s supposed to be? I couldn’t figure that part out for sure.

nefarius, 2023-05-04:
It is supposed to be here `HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Enum\USB\VID_0A12&PID_0001\a&c9c4e92&0&4\Device Parameters` as value `BthPS3PSMPatchEnabled`, example:

[image]

Your key may slightly differ as it depends on the hardware ID of the dongle and the USB port.

InterClaw, 2023-05-04:
Okay, so I found the value here:
`Computer\HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Enum\USB\VID_8087&PID_0029\6&28cf390b&1&13\Device Parameters`

1. It was set to `1`.
2. So I opened the utility and unticked it and closed the utility again.
3. (Now the filter is off until I restart).
4. Refreshed in Registry Editor, but the value was still `1`. Not a good sign.
5. So I manually changed it to `0`.
6. Restart.
7. In the utility, the checkbox is ticked again.
8. In the registry, the value is still on `0`.

So the value was created there at some point. Found no other references to `BthPS3PSMPatchEnabled`. But it's not getting written to and it's not being read from either.

I'm wondering if there is some sort of device mismatch or something. Perhaps the program is expecting this to be at some other location in the registry? There's nothing more to it than just using the included uninstaller, right? Do I manually need to remove some devices or something to fully uninstall it?

InterClaw, 2023-05-26:
Hey  @nefarius any luck with this so far? No rush ofc, just curious what it could be. :) Ready to test whenever.

nefarius, 2023-05-26:
No. If so, the commit history would reflect changes.

nefarius, 2023-07-05:
`devel` branch has all the changes I wanna throw into a new Beta release, almost done so stay tuned.

nefarius, 2023-07-08:
[New pre-release is out, please test](https://github.com/ViGEm/BthPS3/releases/tag/setup-v2.7.296)!

InterClaw, 2023-07-09:
I downloaded and installed v2.7.296 and restarted when it prompted me. (I didn't have the previous version installed.)

When trying to start the utility (admin mode), I'm getting this error message.

[image]

So I uninstalled, rebooted, installed v2.6.174 again, and rebooted. This version is able to start the utility.

nefarius, 2023-07-09:
> I downloaded and installed v2.7.296 and restarted when it prompted me. (I didn't have the previous version installed.)
>
> When trying to start the utility (admin mode), I'm getting this error [...]

Works perfectly over here, first try:

[image]

nefarius, 2023-07-09:
Previous post was Windows 11, here on Windows 10:

[image]

nefarius, 2023-07-09:
And as a special guest; Windows 10 version 1507!

[image]

InterClaw, 2023-07-09:
I'm running this version:

[image]

Is there a way to check if the BthPS3 filter driver actually does get installed, or if it's an access problem I'm having?

nefarius, 2023-07-09:
Well, yeah, see my Screenshots, share what's happening on your end or all we can do is guess and get nowhere.

InterClaw, 2023-07-09:
Ah I didn't realize! 😆 

On v2.7.296 I get no such device showing up. So it doesn't get installed for some reason then.

[image]

When installing the lower version v2.6.174 I do say yes to this prompt btw.

[image]

... and on v2.6.174 I'm getting the device. And it's the new version? Now I'm confused...

[image]

nefarius, 2023-07-09:
So am I, last screenshot looks correct, first one is definitely faulty, no idea what has happened there, also the replace file dialog has never happened to me before so IDK what's going on there either.

nefarius, 2023-07-09:
Welp, in regards to the original issue of this, uh, issue. Look in the System event log for messages originating from BthPS3 components, pretty much everything gets logged now, so I can close this topic for good.

InterClaw, 2023-07-09:
Might be relevant, might not be. On v2.6.174, when I do get the Bluetooth device, the driver details is mentioning two files. This is as it should be?

[image]

InterClaw, 2023-07-09:
> Welp, in regards to the original issue of this, uh, issue. Look in the System event log for messages originating from BthPS3 components, pretty much everything gets logged now, so I can close this topic for good.

Yeah, I wish I could, but if I can't start the utility on v2.7.296, I can't disable the PSM patching. :/

nefarius, 2023-07-09:
That is how it should be and was for years now.

nefarius, 2023-07-09:
> Yeah, I wish I could, but if I can't start the utility on v2.7.296, I can't disable the PSM patching. :/

Well, from the info you provided I can't see what's wrong.

nefarius, 2023-07-09:
> Might be relevant, might not be. On v2.6.174, when I do get the Bluetooth device, the driver details is mentioning two files. This is as it should be?
> 
> [image]

That is how it should be and was for years now.

InterClaw, 2023-07-09:
So the BthPS3PSM.sys file I had is an even lower version, 2.6.167.0 as seen above. This file was left in place when uninstalling.

What I did now was move that file out of there and installed v2.7.296.

... and what do you know, now I get the device and can start the utility. ;)

[image]

I'll see what I can find regarding the original issue now. 👍

InterClaw, 2023-07-09:
Okay, so I'm not that familiar with Event Viewer, but I used "Find..." and searched for "BthPS3" and found that the source "Nefarius BthPS3 Profile Driver" seemed relevant, so I filtered on that.

The four events highlighted ought to be me starting v2.7.296 and unticking "Enable PSM patching". Then after a reboot I went in there again to see that it's reenabled (as expected).

[image]

Below are the details of those for entries. Not that much info to be found though. Should I look for something else in there?

```
Log Name:      System
Source:        Nefarius BthPS3 Profile Driver
Date:          2023-07-09 21:14:21
Event ID:      3
Task Category: None
Level:         Information
Keywords:      
User:          SYSTEM
Computer:      CORTEX
Description:
Host radio HCI major version: 11
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Nefarius BthPS3 Profile Driver" Guid="{37dcd579-e844-4c80-9c8b-a10850b6fac6}" />
    <EventID>3</EventID>
    <Version>0</Version>
    <Level>4</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2023-07-09T19:14:21.4729791Z" />
    <EventRecordID>86630</EventRecordID>
    <Correlation />
    <Execution ProcessID="4" ThreadID="168" />
    <Channel>System</Channel>
    <Computer>CORTEX</Computer>
    <Security UserID="S-1-5-18" />
  </System>
  <EventData>
    <Data Name="Version">11</Data>
  </EventData>
</Event>

Log Name:      System
Source:        Nefarius BthPS3 Profile Driver
Date:          2023-07-09 21:14:21
Event ID:      1
Task Category: None
Level:         Information
Keywords:      
User:          SYSTEM
Computer:      CORTEX
Description:
Driver loaded (obj: 0xffff9d8a760a8e30, status: STATUS_SUCCESS)
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Nefarius BthPS3 Profile Driver" Guid="{37dcd579-e844-4c80-9c8b-a10850b6fac6}" />
    <EventID>1</EventID>
    <Version>0</Version>
    <Level>4</Level>
    <Task>0</Task>
    <Opcode>1</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2023-07-09T19:14:21.1420216Z" />
    <EventRecordID>86629</EventRecordID>
    <Correlation />
    <Execution ProcessID="4" ThreadID="168" />
    <Channel>System</Channel>
    <Computer>CORTEX</Computer>
    <Security UserID="S-1-5-18" />
  </System>
  <EventData>
    <Data Name="DriverObjPtr">0xffff9d8a760a8e30</Data>
    <Data Name="Status">0</Data>
  </EventData>
</Event>

Log Name:      System
Source:        Nefarius BthPS3 Profile Driver
Date:          2023-07-09 21:05:57
Event ID:      3
Task Category: None
Level:         Information
Keywords:      
User:          SYSTEM
Computer:      CORTEX
Description:
Host radio HCI major version: 11
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Nefarius BthPS3 Profile Driver" Guid="{37dcd579-e844-4c80-9c8b-a10850b6fac6}" />
    <EventID>3</EventID>
    <Version>0</Version>
    <Level>4</Level>
    <Task>0</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2023-07-09T19:05:57.8054487Z" />
    <EventRecordID>86515</EventRecordID>
    <Correlation />
    <Execution ProcessID="4" ThreadID="240" />
    <Channel>System</Channel>
    <Computer>CORTEX</Computer>
    <Security UserID="S-1-5-18" />
  </System>
  <EventData>
    <Data Name="Version">11</Data>
  </EventData>
</Event>

Log Name:      System
Source:        Nefarius BthPS3 Profile Driver
Date:          2023-07-09 21:05:57
Event ID:      1
Task Category: None
Level:         Information
Keywords:      
User:          SYSTEM
Computer:      CORTEX
Description:
Driver loaded (obj: 0xffffc20fa4ed3df0, status: STATUS_SUCCESS)
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Nefarius BthPS3 Profile Driver" Guid="{37dcd579-e844-4c80-9c8b-a10850b6fac6}" />
    <EventID>1</EventID>
    <Version>0</Version>
    <Level>4</Level>
    <Task>0</Task>
    <Opcode>1</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2023-07-09T19:05:57.7490502Z" />
    <EventRecordID>86514</EventRecordID>
    <Correlation />
    <Execution ProcessID="4" ThreadID="240" />
    <Channel>System</Channel>
    <Computer>CORTEX</Computer>
    <Security UserID="S-1-5-18" />
  </System>
  <EventData>
    <Data Name="DriverObjPtr">0xffffc20fa4ed3df0</Data>
    <Data Name="Status">0</Data>
  </EventData>
</Event>
```

nefarius, 2023-07-09:
Almost, filter on the profile driver event source:

[image]

Select all, right click, `Save Selected Events...`

[image]

Save the `.evtx` file somewhere, **compress it as zip** and attach it here. Don't leave any events out, I need all of them for the full picture, especially the errors.

PatrykIBM, 2023-07-09:
@nefarius can you maybe trigger uninstaller to remove drivers residue from system as well as registry entries for the program? I had similar problem with the driver + no DS3 was connecting via bluetooth. Only removal of all previous devices + clearing out drivers with driverstorage explorer + removing reg entries allow me to have fully working installation.

InterClaw, 2023-07-09:
Sure! Here are all the events. Had to split it up.

[image]

[532-events.zip](https://github.com/ViGEm/BthPS3/files/11996062/532-events.zip)

(I think the recent error are from when it was not finding the device btw.)

nefarius, 2023-07-09:
> @nefarius can you maybe modify uninstaller to remove drivers residue from system as well as registry entries for the program? I had similar problem with the driver + no DS3 was connecting via bluetooth. Only removal of all previous devices + clearing out drivers with driverstorage explorer + removing reg entries allow me to have fully working installation.

See #79

nefarius, 2023-07-09:
> Sure! Here are all the events. Had to split it up.
> 
> [image]
> 
> [532-events.zip](https://github.com/ViGEm/BthPS3/files/11996062/532-events.zip)
> 
> (I think the recent error are from when it was not finding the device btw.)

Dang, that didn't work. The setup doesn't install the filter manifest for some reason. In the meantime do as follows:

- **Close Event Viewer**
- Download [this file](https://[raw.githubusercontent.com/ViGEm/BthPS3/devel/BthPS3PSM/BthPS3PSM.man](https://raw.githubusercontent.com/ViGEm/BthPS3/devel/BthPS3PSM/BthPS3PSM.man) to e.g. `C:\temp\BthPS3PSM.man`
- Open a PowerShell **as Administrator**
- Run `wevtutil im "C:\temp\BthPS3PSM.man"`
  - If nothing is printed in return, it succeeded
- Restart the machine
- Now redo your tests **but** we need this event source instead:  
  [image]
- You should now get way less events and be able to save them to one file

nefarius, 2023-07-09:
Here is a working sequence from my machine:

- `Stored patch status 1 on device instance USB\VID_0A12&PID_0001\a&c9c4e92&0&4`
- `Retrieved patch status 1 for device instance USB\VID_0A12&PID_0001\a&c9c4e92&0&4`
- `Stored patch status 0 on device instance USB\VID_0A12&PID_0001\a&c9c4e92&0&4`
- `Retrieved patch status 0 for device instance USB\VID_0A12&PID_0001\a&c9c4e92&0&4`

InterClaw, 2023-07-09:
Alright, the command worked.

I restarted, then went into the utility, disabled the PSM patching, restarted again, went in an looked, still enabled. :/ It only mentions patch status 1.

[4-events.zip](https://github.com/ViGEm/BthPS3/files/11996128/4-events.zip)

nefarius, 2023-07-09:
> Alright, the command worked.
> 
> I restarted, then went into the utility, disabled the PSM patching, restarted again, went in an looked, still enabled. :/ It only mentions patch status 1.
> 
> No "stored" events.
> 
> [4-events.zip](https://github.com/ViGEm/BthPS3/files/11996128/4-events.zip)

How exactly do you test, what are you doing? The filter is supposed to store its last setting on unload/shutdown, is that what you're doing?

InterClaw, 2023-07-09:
I open the utility, elevation prompt, click the 2nd tab, untick the checkbox, close the utility, restart.

Then go in and check.

nefarius, 2023-07-09:
Aha! I was just able to reproduce it on one of my machines, and I think I know what's happening, will test a potential fix - tomorrow 😉

nefarius, 2023-07-10:
Alright, new release in the making.

nefarius, 2023-07-10:
Alright, tested latest build, now all looks good. Preparing new release. For future testing, make sure to turn **off** both of these or it will not work as expected:

[image]

[image]

nefarius, 2023-07-10:
Alright @InterClaw please test this one, works on my end now as desired: http://sharex.nefarius.at/u/BthPS3Setup_x64.msi

InterClaw, 2023-07-10:
It works! 🎉 

I first uninstalled v2.7.296, rebooted, and then installed this new version (v2.8.313?). But then I got the problem where it didn't install the bluetooth device for me. Despite BthPS3PSM.sys being version "2.8.313.0".

So then I uninstalled v2.8.313, rebooted, manually deleted BthPS3PSM.sys, and then installed again. And then I got the bluetooth device so I could start the UI.

I disabled the two options as instructed and rebooted again.

The settings remained off! And my controller could connect on the first attempt, as expected now.

I noticed the bluetooth device "Nefarius Bluetooth PS Enumerator" is still there, even when the filter is disabled, but I guess it's not really doing anything right now. 👍

nefarius, 2023-07-10:
> I noticed the bluetooth device "Nefarius Bluetooth PS Enumerator" is still there, even when the filter is disabled, but I guess it's not really doing anything right now. 👍

That is expected behaviour and xorr

PR #87 (2023-07-10): Fix storing and retrieving registry setting on reboot
Fixes #73
commit 650c1146 (2023-07-10): Partially fixes #73
closed 2023-07-10 by nefarius