nefarius/BthPS3 #122: Won't install. Nefarius BthPS3 Bluetooth Drivers Setup ended prematurely

Phalk, 2025-04-18:
Log:
Info: InstallDrivers - USE_MODERN = True
Info: --- BEGIN UninstallDrivers ---
Info: Removing lower filter entry
Info: Removed lower filter entry
Info: Disabling BthPS3 service
Info: Disabled BthPS3 service
Info: Disabling radio
Info: Disabled radio
Info: Restarting radio device
Info: Restarted radio device
Info: Waiting for radio device to come online...
Info: Radio arrival event, path: \\?\USB#VID_0A5C&PID_2190#001A7D0AC490#{92383b0e-f90e-4ac9-8d44-8c2d0d0ebda2}
Info: Radio available after restart
Info: Removing driver package C:\WINDOWS\System32\DriverStore\FileRepository\bthps3psm.inf_amd64_5699aec9517e2633\bthps3psm.inf
Info: Enabling radio
Info: Enabled radio
Info: --- END UninstallDrivers ---
Info: --- BEGIN InstallDrivers ---
Info: installDir = C:\Program Files\Nefarius Software Solutions\BthPS3\
Info: driversDir = C:\Program Files\Nefarius Software Solutions\BthPS3\drivers
Info: nefconDir = C:\Program Files\Nefarius Software Solutions\BthPS3\nefcon
Info: archShortName = X64
Info: nefconcPath = C:\Program Files\Nefarius Software Solutions\BthPS3\nefcon\X64\nefconc.exe
Info: bthPs3DriverDir = C:\Program Files\Nefarius Software Solutions\BthPS3\drivers\BthPS3_X64
Info: bthPs3PsmDriverDir = C:\Program Files\Nefarius Software Solutions\BthPS3\drivers\BthPS3PSM_X64
Info: bthPs3InfPath = C:\Program Files\Nefarius Software Solutions\BthPS3\drivers\BthPS3_X64\BthPS3.inf
Info: bthPs3PsmInfPath = C:\Program Files\Nefarius Software Solutions\BthPS3\drivers\BthPS3PSM_X64\BthPS3PSM.inf
Info: bthPs3NullInfPath = C:\Program Files\Nefarius Software Solutions\BthPS3\drivers\BthPS3_X64\BthPS3_PDO_NULL_Device.inf
Info: BthPS3PSM command stdout: 2025-04-18T22:04:50Z ERROR Failed to install INF file, error: InstallHinfSectionW failed with error: Não foi possível encontrar o módulo especificado.
 (126)

Info: BthPS3PSM command stderr: 
Info: Filter installer failed with exit code: 126, message: Não foi possível encontrar o módulo especificado.
Info: --- END InstallDrivers FAILURE ---
Info: CustomAction InstallDrivers returned actual error code 1603 (note this may not be 100% accurate if translation happened inside sandbox)
Info: Action ended 18:04:50: InstallFinalize. Return value 3.
CommonData: Message type: 2, Argument: 0
CommonData: Message type: 2, Argument: 0
CommonData: Message type: 0, Argument: 1033, 1252
CommonData: Message type: 1, Argument: Nefarius BthPS3 Bluetooth Drivers
ActionStart: Action 18:04:50: Rollback. Rolling back action: [1]
ActionData: 1: InstallDrivers 
ActionData: 1: Copying new files File: [1],  Directory: [9],  Size: [6] 
ActionData: 1: Creating folders Folder: [1] 
ActionData: 1: Updating component registration 
CommonData: Message type: 2, Argument: 1
CommonData: Message type: 2, Argument: 1
Info: Action ended 18:04:50: INSTALL. Return value 3.
InstallEnd: 1: Nefarius BthPS3 Bluetooth Drivers 2: {CC32A6ED-BDFE-4D51-9FFF-2AB53EAEDE18} 3: 3 
CommonData: 1: 2 2: 0 
CommonData: 1: 2 2: 1 
Info: Action ended 18:04:50: INSTALL. Return value 3.

Phalk, 2025-04-18:
Used to work before. This is the newer release install.

nefarius, 2025-04-18:
Use the Legacy mode, it's there for that exact reason.

nefarius, 2025-04-18:
Oh hang on, this is new indeed:

```
Info: BthPS3PSM command stdout: 2025-04-18T22:04:50Z ERROR Failed to install INF file, error: InstallHinfSectionW failed with error: Não foi possível encontrar o módulo especificado.
```
`The specified module could not be found.`the `BthPS3PSM.inf` failed to install 🤔

nefarius, 2025-04-18:
@Phalk attach the file `C:\Windows\INF\setupapi.dev.log` from the affected machine.

nefarius, 2025-04-18:
Also what exact Windows version are you on. The issue template is there for a reason, having to ask these basics every time is quite annoying.

Phalk, 2025-04-18:
-

[setupapi.app.log](https://github.com/user-attachments/files/19817129/setupapi.app.log)

 [ X ] I have searched open and closed issues for duplicates
- [ X ] I am submitting a bug report for existing functionality that does not work as intended
- [ X ] I have read [the project documentation](https://docs.nefarius.at/projects/) and not found a solution to my problem there
- [ X ] This isn't a feature request or a discussion topic

----------------------------------------

### Bug description

Describe here the issue that you are experiencing.

### Steps to reproduce

- tried to execute the installer (a previous version was already installed, which doesn't work anymore)
- started the process
- error

**Actual result:** The original post.
**Expected result:** Successful installation

### Screenshots

-

### Machine info

**CPU-Architecture:** PC, i5-3570k x64
**Windows version:** Windows 11 23H2
**Software/driver version(s):** Latest

nefarius, 2025-04-18:
Wrong file. Please read **carefully**.

closed 2025-04-26 by nefarius